pal-robotics/tiago_simulation #12: Change joint state controller

huiwenzhang, 2018-06-19:
Hi, 
I am using your tiago simulation tacks in ROS indigo, Ubuntu 14.04. I found the published joint state is wrong sometimes, as posted by @jacknlliu in [issue in ros_control](https://github.com/ros-controls/ros_controllers/issues/317). The joint state return by joint_state_controller can be very big, e.g. the `arm_5_joint` returns joint position above 4 rad. However, the joint position limits is [-2.09, 2.09]. Apparently, the joint information is wrong. @jacknlliu declared that using `joint_state_publisher` plugin in gazebo can solve the problem. So I am trying to do it. Basically what I have done is concluded as follows:

1. **Disable the default `joint_state_controller`**．To  do this, I made some changes as follows:
https://github.com/huiwenzhang/tiago_simulation/blob/fbde8a0016ed1689874642e2acd02aa8107897ea/tiago_controller_configuration_gazebo/launch/default_controllers.launch#L16 (comment this)
2. **Add joint_state_publisher plugin here:**
https://github.com/huiwenzhang/tiago_robot/blob/78482ab88bccd23e3c5c4960e1c8520b648dcbcc/tiago_description/gazebo/gazebo.urdf.xacro#L47-L61
I created another file for titanium robot, because they have different joints as shown in plugin settings (I am not sure should I list all of the movable joints of the robot)
3. **Load the corresponding xarco file**
https://github.com/huiwenzhang/tiago_robot/blob/78482ab88bccd23e3c5c4960e1c8520b648dcbcc/tiago_description/robots/tiago_titanium.urdf.xacro#L30

I want to know is there anything else I missed? More importantly, whether the above changes are correct and enough for my purpose.

Thanks

bmagyar, 2018-06-19:
The Tiago launch files mirror the same system that is provided with the real robot platform. The issue on `ros_control` was reported but without proper diagnosis or reproducibility. I'd still recommend investigating there so we can provide a proper fix, if one is needed.

huiwenzhang, 2018-06-19:
Yep, you're right. I will keep tracking the issue on ros_control and probably help to formulate the problem with repeatable examples.

closed 2023-04-20 by saikishor