pal-robotics/tiago_simulation #13: Tiago doesn't move

FCam1, 2019-04-10:
Hi,
My config: ubuntu16.04 - ROS kinetic.  

I try to follow the [Tiago tutorials](http://wiki.ros.org/Robots/TIAGo/Tutorials) but I meet an issue when I launch gazebo with:
 `roslaunch tiago_gazebo tiago_gazebo.launch public_sim:=true robot:=titanium world:=simple_office_with_people`
The robot keep his head and his arm 'soft' so I think to a **problem of controllers**. And also the robot doesn't move with the `key_teleop` command. 

<p align="center">
[image]
</p>

Moreover I have a **FATAL error** maybe linked : 
```
[FATAL] [1554900139.882791288, 0.001000000]: Failed to create robot simulation
interface loader: Could not find library corresponding to plugin
pal_hardware_gazebo/PalHardwareGazebo.Make sure the plugin description XML 
file has the   correct name of the library and that the library actually exists.  
```
Yet in my worksape I have the plugin .xml: `tiago_public_ws/src/pal_hardware_gazeb/pal_hardware_gazebo_plugins.xml` 

Could you give me ideas to debug?

v-lopez, 2019-04-10:
Hi @FCam1,

Sorry to hear that you are having issues with this.

Could you post the full logs of the `tiago_gazebo.launch`?
Also, make sure that the full workspace was built properly, that  you have sourced it, and that you can find the gazebo ros control plugins using `rospack plugins --attrib=plugin gazebo_ros_control`.

Finally, we've seen some interactions if people have ros control installed from ROS sources, you should only have our version usable from the workspace. As discussed here: https://answers.ros.org/question/292578/is-the-tutorial-to-compile-tiago-with-ros-kinetic-working/

It may manifest as an runtime issue, instead of a compilation problem.

FCam1, 2019-04-10:
Thanks for answering

`tiago_gazebo.launch` gives me this :
```
.. logging to /home/fcaminad/.ros/log/c5b87abe-5ba2-11e9-9582-f8b156bf986d/roslaunch-bandaisan-3134.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

deprecated: xacro tags should be prepended with 'xacro' xml namespace.
Use the following script to fix incorrect usage:
        find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'
when processing file: /local/fcaminad/devel/tiago_public_ws/src/hey5_description/urdf/deg_to_rad.urdf.xacro
included from: /local/fcaminad/devel/tiago_public_ws/src/hey5_description/urdf/hey5_hand.urdf.xacro
included from: /local/fcaminad/devel/tiago_public_ws/src/tiago_robot/tiago_description/urdf/end_effector/end_effector.urdf.xacro
included from: /local/fcaminad/devel/tiago_public_ws/src/tiago_robot/tiago_description/robots/tiago.urdf.xacro

started roslaunch server http://bandaisan:42507/

SUMMARY
========

PARAMETERS
 * /arm_controller/constraints/arm_1_joint/goal: 0.02
 * /arm_controller/constraints/arm_2_joint/goal: 0.02
 * /arm_controller/constraints/arm_3_joint/goal: 0.02
 * /arm_controller/constraints/arm_4_joint/goal: 0.02
 * /arm_controller/constraints/arm_5_joint/goal: 0.02
 * /arm_controller/constraints/arm_6_joint/goal: 0.02
 * /arm_controller/constraints/arm_7_joint/goal: 0.02
 * /arm_controller/constraints/goal_time: 0.6
 * /arm_controller/constraints/stopped_velocity_tolerance: 5.0
 * /arm_controller/joints: ['arm_1_joint', '...
 * /arm_controller/type: position_controll...
 * /collision_model_parameters/blacklist/default: [{'second': 'tors...
 * /collision_model_parameters/links_with_padding: ['arm_4_link', 'a...
 * /collision_model_parameters/padding: 0.025
 * /force_torque/wrist_ft/frame: wrist_ft_link
 * /force_torque/wrist_ft/sensor_joint: arm_7_joint
 * /force_torque_sensor_controller/publish_rate: 50
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /gains/FFJ1/d: 0
 * /gains/FFJ1/i: 10
 * /gains/FFJ1/p: 100
 * /gains/FFJ2/d: 0
 * /gains/FFJ2/i: 10
 * /gains/FFJ2/p: 100
 * /gains/FFJ3/d: 0
 * /gains/FFJ3/i: 10
 * /gains/FFJ3/p: 100
 * /gains/FFJ4/d: 0
 * /gains/FFJ4/i: 10
 * /gains/FFJ4/p: 100
 * /gains/MFJ1/d: 0
 * /gains/MFJ1/i: 10
 * /gains/MFJ1/p: 100
 * /gains/MFJ2/d: 0
 * /gains/MFJ2/i: 10
 * /gains/MFJ2/p: 100
 * /gains/MFJ3/d: 0
 * /gains/MFJ3/i: 10
 * /gains/MFJ3/p: 100
 * /gains/MFJ4/d: 0
 * /gains/MFJ4/i: 10
 * /gains/MFJ4/p: 100
 * /gains/RFJ1/d: 0
 * /gains/RFJ1/i: 10
 * /gains/RFJ1/p: 100
 * /gains/RFJ2/d: 0
 * /gains/RFJ2/i: 10
[... 815 lines cut ...]
[WARN] [1554908760.407726, 113.745000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1554908760.531721, 113.867000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[default_controllers_spawner-5] process has finished cleanly
log file: /home/fcaminad/.ros/log/d3e17b5a-5ba1-11e9-a6ae-f8b156bf986d/default_controllers_spawner-5*.log
[hand_controller_spawner-9] process has finished cleanly
log file: /home/fcaminad/.ros/log/d3e17b5a-5ba1-11e9-a6ae-f8b156bf986d/hand_controller_spawner-9*.log
```
---
I source with `source /local/fcaminad/devel/tiago_public_ws/src/setup.bash`

The result of the `rospack plugins --attrib=plugin gazebo_ros_control` seems OK: 
```
pal_hardware_gazebo /local/fcaminad/devel/tiago_public_ws/src/pal_hardware_gazebo/pal_hardware_gazebo_plugins.xml
gazebo_ros_control /local/fcaminad/devel/tiago_public_ws/src/gazebo_ros_pkgs/gazebo_ros_control/robot_hw_sim_plugins.xml
```
---
The result of the `catkin build -DCATKIN_ENABLE_TESTING=0` seems OK:
``` python
[build] Summary: All 121 packages succeeded!                                   
[build]   Ignored:   11 packages were skipped or are blacklisted.              
[build]   Warnings:  22 packages succeeded with warnings.                      
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 4 minutes and 23.7 seconds total.
```

v-lopez, 2019-04-11:
It's weird, there are nodes not being found:

> ERROR: cannot launch node of type [head_action/head_action]: can't locate node [head_action]
> ERROR: cannot launch node of type [joy_teleop/incrementer_server]: can't locate node [incrementer_server] in package [joy_teleop]

But are on the kinetic rosinstall file: https://github.com/pal-robotics/tiago_tutorials/blob/kinetic-devel/tiago_public.rosinstall

Please make sure you have the latest rosinstall file, that all your repositories are up to date, and run catkin build with `-s` so we can see a summary of what's being compiled and what's being ignored.

FCam1, 2019-04-11:
In a first time I started over from scratch:
I checked my [install and dependencies](http://wiki.ros.org/Robots/TIAGo/Tutorials/Installation/TiagoSimulation), cleaned my catkin workspace,  made a new rosinstall and a `catkin build -DCATKIN_ENABLE_TESTING=0 -s`   but I still had errors.
Logs during build:
```
[build] Successful packages:                                                   
                                                                               
 [Successful] aruco                                                            
 [Successful] aruco_msgs                                                       
 [Successful] aruco_ros                                                        
 [Successful] backward_ros                                                     
 [Successful] combined_robot_hw                                                
 [Successful] combined_robot_hw_tests                                          
 [Successful] controller_interface                                             
 [Successful] controller_manager                                               
 [Successful] controller_manager_msgs                                          
 [Successful] controller_manager_tests                                         
 [Successful] ddynamic_reconfigure_python                                      
 [Successful] demo_motions                                                     
 [Successful] diff_drive_controller                                            
 [Successful] dynamic_introspection                                            
 [Successful] effort_controllers                                               
 [Successful] force_torque_sensor_controller                                   
 [Successful] forward_command_controller                                       
 [Successful] four_wheel_steering_controller                                   
 [Successful] gazebo_dev                                                       
 [Successful] gazebo_msgs                                                      
 [Successful] gazebo_plugins                                                   
 [Successful] gazebo_ros                                                       
 [Successful] gazebo_ros_control                                               
 [   Ignored] gazebo_ros_pkgs                                                  
 [Successful] gripper_action_controller                                        
 [Successful] hardware_interface                                               
 [Successful] head_action                                                      
 [Successful] hey5_description                                                 
 [Successful] imu_sensor_controller                                            
 [Successful] joint_limits_interface                                           
 [Successful] joint_state_controller                                           
 [Successful] joint_torque_sensor_state_controller                             
 [Successful] joint_trajectory_controller                                      
 [Successful] joy_teleop                                                       
 [Successful] key_teleop                                                       
 [Successful] look_hand                                                        
 [Successful] look_to_point                                                    
 [Successful] mode_state_controller                                            
 [Successful] mouse_teleop                                                     
 [Successful] pal_behaviour_msgs                                               
 [Successful] pal_carbon_collector                                             
 [Successful] pal_common_msgs                                                  
 [Successful] pal_control_msgs                                                 
 [Successful] pal_detection_msgs                                               
 [Successful] pal_device_msgs                                                  
 [Successful] pal_gazebo_plugins                                               
 [Successful] pal_gazebo_worlds                                                
 [Successful] pal_gripper                                                      
 [Successful] pal_gripper_controller_configuration                             
 [Successful] pal_gripper_controller_configuration_gazebo                      
 [Successful] pal_gripper_description                                          
 [Successful] pal_gripper_gazebo                                               
 [Successful] pal_hardware_gazebo                                              
 [Successful] pal_hardware_interfaces                                          
 [Successful] pal_interaction_msgs                                             
 [Successful] pal_motion_model_msgs                                            
 [   Ignored] pal_msgs                                                         
 [Successful] pal_multirobot_msgs                                              
 [Successful] pal_navigation_msgs                                              
 [Successful] pal_navigation_sm                                                
 [Successful] pal_parallel_gripper_wrapper                                     
 [Successful] pal_python                                                       
[... cut ...]
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                       
```  
After that I decided to use [Robotpkg](http://robotpkg.openrobots.org/debian.html). Thanks to @olivier-stasse.
So I installed via apt: `sudo apt install robotpkg-tiago-simulation`
The basic simulation works, I can control Tiago with joystick. Now I am going to continue follow the tutorials. 
Thanks @v-lopez   for your help

saikishor, 2023-04-12:
Hello,

Closing the issue due to inactivity.

Best Regards,
PAL Robotics Team.

closed 2023-04-12 by saikishor