pal-robotics/tiago_simulation #15: No module named 'launch_pal'

fgonzalezr1998, 2021-03-08:
Hello, I am trying to launch the Tiago simulation in ROS2 Foxy. I have done a ``git clone --recursive -b foxy-devel https://github.com/pal-robotics/tiago_simulation`` in my workspace and it compile well but, when I run ``ros2 launch tiago_gazebo tiago_gazebo.launch.py``, the output is the following:
```
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2021-03-08-11-25-45-065262-ubuntu20-8837
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=InvalidLaunchFileError('py')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 62, in get_launch_description_from_python_launch_file
    launch_file_module = load_python_launch_file_as_module(python_launch_file_path)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 37, in load_python_launch_file_as_module
    loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/ubuntu/tiago_foxy_ws/install/tiago_gazebo/share/tiago_gazebo/launch/tiago_gazebo.launch.py", line 37, in <module>
    from launch_pal.include_utils import include_launch_py_description
ModuleNotFoundError: No module named 'launch_pal'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 125, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [py]: No module named 'launch_pal'
```

How can I solve my problem? Thank you for all

v-lopez, 2021-03-08:
Hi, this is still work in progress, we hope to finish, release and write some tutorials in the upcoming weeks.

fgonzalezr1998, 2021-03-08:
OK, Thank you! @v-lopez

miccol, 2021-06-22:
Is there any update on this issue? I am having the same problem.

SteveMacenski, 2021-06-22:
https://github.com/pal-robotics/launch_pal did you build this in your workspace?

v-lopez, 2021-06-23:
Yeah we need to add it as a dependency on `tiago_description` our bad, also it is released in foxy, so you could install `ros-foxy-launch-pal`.

v-lopez, 2021-06-23:
As an update, we're almost ready to release pmb2 on foxy, for tiago we're waiting on some MoveIt! and ros_control releases first. Hopefully during the summer

Sjokoladen, 2021-09-08:
@fgonzalezr1998  How did you build your workspace properly? Also, did you have that tutorial @v-lopez ?

saikishor, 2022-12-12:
Hello @Dharnish4,

> @v-lopez @SteveMacenski Is there any update on ROS2 pal tutorials. I am trying to use nav2 now for my project. Only way to do that is use turtlebot but I am working on tiago robot. Any info on how I can use tiago simulation with nav2 would be really appreciated.

Please kindly take a look at the README here: https://github.com/pal-robotics/tiago_simulation/tree/humble-devel. 

Best Regards,
PAL Robotics Team.

closed 2022-12-12 by saikishor