pal-robotics/tiago_simulation #19: [ROS2] controller_manager not available

AndyZe, 2022-09-12:
I'm trying to do some mobile manipulation simulations in ROS2 and was hoping to use Tiago. But, it looks like ROS2 launching isn't functional yet. Is that correct?

The main issue I see is that the `controller_manager` is not launched:

```
[spawner-8] [INFO] [1662988351.731856152] [spawner_head_controller]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1662988351.753627861] [spawner_torso_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1662988351.753633951] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-10] [ERROR] [1662988353.659231753] [spawner_hand_controller]: Controller manager not available
[spawner-5] [ERROR] [1662988353.729551282] [spawner_mobile_base_controller]: Controller manager not available
[spawner-9] [ERROR] [1662988353.739216180] [spawner_arm_controller]: Controller manager not available
[spawner-8] [ERROR] [1662988353.739462857] [spawner_head_controller]: Controller manager not available
```

Can anybody describe what state this repository is in for ROS2? `Working` / `expected to be working soon` / `not anytime soon`

saikishor, 2022-09-12:
Hello @AndyZe,

We are currently working towards integration of the ROS2 on our TIAGo robot. it is expected to be working sometime in near future. I hope this information helps.

Thank you,

Best Regards,
PAL Robotics Team.

AndyZe, 2022-09-12:
Thanks! If you can, please ping me here when it is functional. There's the potential to be featured in some cool features/simulations, maybe even at ROSCon 2023.

Noel215, 2022-09-22:
Hi @AndyZe, TIAGo simulation and navigation are already working for the latest distro of ROS2, humble.

AndyZe, 2022-10-03:
Thanks @Noel215. Can you give me any details on how to install/ launch it? I've tried `ros2 launch tiago_gazebo tiago_gazebo.launch.py` but get a bunch of "Controller manager not available" errors.

Noel215, 2022-10-03:
Of course I can @AndyZe. I'll explain you shortly, but I'm preparing a readme file with more detailed instructions.

First of all I recommend you changing to Cyclone DDS. With the Fast-DDS we are having some issues, similar that the one you reported. You can do it by setting a environment variable with the command `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`.

Then you can use the [tiago_public.rosinstall](https://github.com/pal-robotics/tiago_tutorials/blob/humble-devel/tiago_public.rosinstall) to get all the packages needed for TIAGo simulation. Now we've also included moveit2 with the simulation!

Finally you can launch the simulation with the command you've been using `ros2 launch tiago_gazebo tiago_gazebo.launch.py`

AndyZe, 2022-10-06:
I got it working. I am on Rolling and the key seems to have been `sudo apt update / upgrade`. Thanks!

(I'll probably switch to Humble soon)

closed 2022-10-06 by AndyZe