pal-robotics/tiago_simulation #20: Unused args

gerardcanal, 2023-01-31:
Hello,
When running `roslaunch tiago_2dnav_gazebo tiago_navigation.launch public_sim:=true`

I get a
`
RLException: unused args [multiple, robot_namespace, is_sim] for include of [/opt/tiago_public_ws/src/pal_navigation_cfg_public/pal_navigation_cfg_tiago/launch/state_machine.launch]
`

Which apparently comes from [state_machine.launch](https://github.com/pal-robotics/pal_navigation_cfg_public/blob/master/pal_navigation_cfg_tiago/launch/state_machine.launch) which doesn't have those arg,s but they are set in [navigation.launch](https://github.com/pal-robotics/tiago_simulation/blob/74f770757725a164335be78f044362d050a1dc22/tiago_2dnav_gazebo/launch/navigation.launch#L36)

I see some commits from yesterday, not sure if that broke it or if it was broken from before. This only happens when using public_sim.

saikishor, 2023-01-31:
Hello @gerardcanal,

Thank you for pointing out the issue. Tomorrow morning, we will try to review the issue and get back to you. 

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2023-02-01:
Hello @gerardcanal ,

Thank you for reporting the issue. We have updated the repositories with the new fixes. Please pull the new changes from the repositories to test them.

Thank you,

Best Regards,
PAL Robotics Team.

gerardcanal, 2023-02-01:
Thanks Sai!
It seems to work now.