pal-robotics/tiago_simulation #21: RViz doesn't display Tiago's depth image

jmguerreroh, 2023-03-01:
The image published on the topic `/head_front_camera/depth_registered/image_raw` cannot be displayed in RViz

[image]

saikishor, 2023-03-02:
Hello @jmguerreroh,

Please kindly provide us with more info so we could reproduce the issue. on TIAGo melodic and noetic, you can get the depth image from the topic : `/xtion/depth_registered/image_raw`.

Thank you,

Best Regards,
PAL Robotics Team.

jmguerreroh, 2023-03-02:
Hello @saikishor ,

I am using Ubuntu 22.04 and ROS Humble distro. Topic `/xtion/depth_registered/image_raw` does not exists due its occurs in the simulated TIAGo, so the topic is `/head_front_camera/depth_registered/image_raw` and depth image in RViz is dark.

After different times or head movements, the image is displayed, but not always.

Thank you.

closed 2023-04-12 by saikishor