pal-robotics/tiago_simulation #22: Tiago robot does not move any joints

nilutpolkashyap, 2023-03-12:
### Issue: Tiago Simulation ROS 2 does not move any of its joints 

I was trying to follow the ROS 2 Tiago simulation steps from [TIAGo ROS 2 Simulation](https://github.com/pal-robotics/tiago_simulation/tree/humble-devel) but the robot does not move any joints due to 

I launched the simulation using  the command:
```ros2 launch tiago_gazebo tiago_gazebo.launch.py [arm:=no-arm]```

**Gazebo Output**
[image]

**Terminal Output I get:**
```
[INFO] [launch]: All log files can be found below /home/nilutpolk/.ros/log/2023-03-12-22-37-22-304938-nilutpolk-Predator-PH315-52-9171
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [play_motion2_node-10]: process started with pid [9197]
[INFO] [tuck_arm.py-15]: process started with pid [9212]
[INFO] [gzserver-1]: process started with pid [9173]
[INFO] [gzclient-2]: process started with pid [9175]
[INFO] [spawn_entity.py-3]: process started with pid [9177]
[INFO] [spawner-4]: process started with pid [9179]
[INFO] [spawner-5]: process started with pid [9181]
[INFO] [spawner-6]: process started with pid [9184]
[INFO] [spawner-7]: process started with pid [9187]
[INFO] [spawner-8]: process started with pid [9190]
[INFO] [spawner-9]: process started with pid [9193]
[INFO] [twist_mux-11]: process started with pid [9199]
[INFO] [twist_marker-12]: process started with pid [9201]
[INFO] [joy_teleop-13]: process started with pid [9203]
[INFO] [robot_state_publisher-14]: process started with pid [9207]
[twist_mux-11] [INFO] [1678640845.477675986] [twist_mux]: Topic handler 'topics.joystick' subscribed to topic 'joy_vel': timeout = 0.500000s , priority = 100.
[twist_mux-11] [INFO] [1678640845.484809285] [twist_mux]: Topic handler 'topics.keyboard' subscribed to topic 'key_vel': timeout = 0.500000s , priority = 90.
[twist_mux-11] [INFO] [1678640845.486246950] [twist_mux]: Topic handler 'topics.marker' subscribed to topic 'marker_vel': timeout = 0.500000s , priority = 99.
[twist_mux-11] [INFO] [1678640845.486491947] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'nav_vel': timeout = 0.500000s , priority = 10.
[twist_mux-11] [INFO] [1678640845.486723856] [twist_mux]: Topic handler 'topics.phone' subscribed to topic 'phone_vel': timeout = 0.500000s , priority = 98.
[twist_mux-11] [INFO] [1678640845.486926619] [twist_mux]: Topic handler 'topics.rviz' subscribed to topic 'rviz_joy_vel': timeout = 0.500000s , priority = 95.
[twist_mux-11] [INFO] [1678640845.487172227] [twist_mux]: Topic handler 'topics.servoing' subscribed to topic 'servoing_cmd_vel': timeout = 0.500000s , priority = 20.
[twist_mux-11] [INFO] [1678640845.487441914] [twist_mux]: Topic handler 'topics.tablet' subscribed to topic 'tab_vel': timeout = 0.500000s , priority = 100.
[twist_mux-11] [INFO] [1678640845.487794324] [twist_mux]: Topic handler 'locks.joystick' subscribed to topic 'joy_priority': timeout = None , priority = 100.
[twist_mux-11] [INFO] [1678640845.489227362] [twist_mux]: Topic handler 'locks.loop_closure' subscribed to topic 'stop_closing_loop': timeout = None , priority = 200.
[twist_mux-11] [INFO] [1678640845.489880701] [twist_mux]: Topic handler 'locks.pause' subscribed to topic 'pause_navigation': timeout = None , priority = 100.
[robot_state_publisher-14] [INFO] [1678640845.879289646] [robot_state_publisher]: got segment arm_1_link
[robot_state_publisher-14] [INFO] [1678640845.880836298] [robot_state_publisher]: got segment arm_2_link
[robot_state_publisher-14] [INFO] [1678640845.881005574] [robot_state_publisher]: got segment arm_3_link
[robot_state_publisher-14] [INFO] [1678640845.881039348] [robot_state_publisher]: got segment arm_4_link
[robot_state_publisher-14] [INFO] [1678640845.881064384] [robot_state_publisher]: got segment arm_5_link
[robot_state_publisher-14] [INFO] [1678640845.881079872] [robot_state_publisher]: got segment arm_6_link
[robot_state_publisher-14] [INFO] [1678640845.881094980] [robot_state_publisher]: got segment arm_7_link
[robot_state_publisher-14] [INFO] [1678640845.881114549] [robot_state_publisher]: got segment arm_tool_link
[robot_state_publisher-14] [INFO] [1678640845.881175180] [robot_state_publisher]: got segment base_antenna_left_link
[robot_state_publisher-14] [INFO] [1678640845.881202838] [robot_state_publisher]: got segment base_antenna_right_link
[robot_state_publisher-14] [INFO] [1678640845.881225620] [robot_state_publisher]: got segment base_cover_link
[robot_state_publisher-14] [INFO] [1678640845.881251265] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-14] [INFO] [1678640845.881274950] [robot_state_publisher]: got segment base_imu_link
[robot_state_publisher-14] [INFO] [1678640845.881297776] [robot_state_publisher]: got segment base_laser_link
[robot_state_publisher-14] [INFO] [1678640845.881323460] [robot_state_publisher]: got segment base_link
[robot_state_publisher-14] [INFO] [1678640845.881674763] [robot_state_publisher]: got segment base_mic_back_left_link
[robot_state_publisher-14] [INFO] [1678640845.884275759] [robot_state_publisher]: got segment base_mic_back_right_link
[robot_state_publisher-14] [INFO] [1678640845.884354867] [robot_state_publisher]: got segment base_mic_front_left_link
[robot_state_publisher-14] [INFO] [1678640845.884370145] [robot_state_publisher]: got segment base_mic_front_right_link
[robot_state_publisher-14] [INFO] [1678640845.884380078] [robot_state_publisher]: got segment base_sonar_01_link
[robot_state_publisher-14] [INFO] [1678640845.884390637] [robot_state_publisher]: got segment base_sonar_02_link
[robot_state_publisher-14] [INFO] [1678640845.884403352] [robot_state_publisher]: got segment base_sonar_03_link
[robot_state_publisher-14] [INFO] [1678640845.884420802] [robot_state_publisher]: got segment caster_back_left_1_link
[robot_state_publisher-14] [INFO] [1678640845.884457310] [robot_state_publisher]: got segment caster_back_left_2_link
[robot_state_publisher-14] [INFO] [1678640845.884470701] [robot_state_publisher]: got segment caster_back_right_1_link
[robot_state_publisher-14] [INFO] [1678640845.884481755] [robot_state_publisher]: got segment caster_back_right_2_link
[robot_state_publisher-14] [INFO] [1678640845.885631183] [robot_state_publisher]: got segment caster_front_left_1_link
[robot_state_publisher-14] [INFO] [1678640845.885660998] [robot_state_publisher]: got segment caster_front_left_2_link
[robot_state_publisher-14] [INFO] [1678640845.885674754] [robot_state_publisher]: got segment caster_front_right_1_link
[robot_state_publisher-14] [INFO] [1678640845.885684729] [robot_state_publisher]: got segment caster_front_right_2_link
[... 327 lines cut ...]
[tuck_arm.py-15] [ERROR] [1678640962.763693175] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-10] [ERROR] [1678640965.767536692] [play_motion2]: There are no active JointTrajectory controllers available
[tuck_arm.py-15] [ERROR] [1678640965.768528621] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-10] [ERROR] [1678640968.772486630] [play_motion2]: There are no active JointTrajectory controllers available
[tuck_arm.py-15] [ERROR] [1678640968.775665671] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-10] [ERROR] [1678640971.779988979] [play_motion2]: There are no active JointTrajectory controllers available
[tuck_arm.py-15] [ERROR] [1678640971.782472584] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-10] [ERROR] [1678640974.790288898] [play_motion2]: There are no active JointTrajectory controllers available
[tuck_arm.py-15] [ERROR] [1678640974.797674701] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-10] [ERROR] [1678640977.800631737] [play_motion2]: There are no active JointTrajectory controllers available
[tuck_arm.py-15] [ERROR] [1678640977.803430413] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-10] [ERROR] [1678640980.809283373] [play_motion2]: There are no active JointTrajectory controllers available
[tuck_arm.py-15] [ERROR] [1678640980.812620229] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-10] [ERROR] [1678640983.817831529] [play_motion2]: There are no active JointTrajectory controllers available
[tuck_arm.py-15] [ERROR] [1678640983.824052260] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-10] [ERROR] [1678640986.828474752] [play_motion2]: There are no active JointTrajectory controllers available
[tuck_arm.py-15] [ERROR] [1678640986.832577411] [arm_tucker]: play_motion2 is not ready
[gzserver-1] terminate called after throwing an instance of 'std::runtime_error'
[gzserver-1]   what():  Error in state! Internal state_handle is NULL.
[ERROR] [gzserver-1]: process has died [pid 9173, exit code -6, cmd 'gzserver -s libgazebo_ros_init.so -s libgazebo_ros_factory.so /home/nilutpolk/bookros2_ws/install/pal_gazebo_worlds/share/pal_gazebo_worlds/worlds/pal_office.world'].
```

**Additional Information**
Version: ROS 2 Humble 
OS: Ubuntu 22.04

Noel215, 2023-03-13:
Hi @nilutpolkashyap,

Please, see https://github.com/pal-robotics/pmb2_simulation/issues/4, it's the same issue.

nilutpolkashyap, 2023-03-13:
Thanks a lot @Noel215 

Compiling the gazebo_ros2_control humble branch and ros2_control humble branch in my local workspace did solve my problem.

closed 2023-03-13 by saikishor