pal-robotics/tiago_simulation #24: ROS 1 example on humble-devel branch

jkaniuka, 2023-05-04:
Hi :wave:,

 I noticed that on the **humble-devel** branch there is a file [`grasp_demo.py`](https://github.com/pal-robotics/tiago_simulation/blob/humble-devel/tiago_gazebo/scripts/grasp_demo.py), which is an example that uses ROS 1. However, the second example in that directory ([`tuck_arm.py`](https://github.com/pal-robotics/tiago_simulation/blob/humble-devel/tiago_gazebo/scripts/tuck_arm.py)) does already use ROS 2. I suspect that this is just an oversight, but it may be confusing to beginners that on the branch regarding ROS 2 there is an example that uses ROS 1. 

Regards,
Jan

saikishor, 2023-05-05:
Hello @jkaniuka,

Thank you for writing to us. We have the script as a part of the migration that needs to be done. However, when you compile the package, the part of the grasp_demo.py script is not installed as per the install rules defined in the [CMakeLists.txt](https://github.com/pal-robotics/tiago_simulation/blob/humble-devel/tiago_gazebo/CMakeLists.txt#L9-L12). At the same time, we understand this might be confusing for beginners. We will consider doing a cleanup soon.

Thank you,

Best Regards,
PAL Robotics Team.

jkaniuka, 2023-05-05:
Great, good luck on the migration to ROS 2 :wink:

closed 2023-05-05 by jkaniuka