pal-robotics/tiago_simulation #26: tiago_gazebo.launch.py /scan_raw missing on simulation

phil123456, 2023-09-11:
Following Rico's book - page 43 - (https://github.com/fmrico/book_ros2)

[Ubuntu 20.04.6 LTS / Foxy]

I try to run **Tiago sim** and drive it around a world map with the **teleop launcher**

I run the **simulation** like this

    cd bookros2_ws/src/
    source install/setup.sh
    ros2 launch br2_tiago sim.launch.py 

I run the **teleop** like this

    cd bookros2_ws/src/
    source install/setup.sh
    ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -r cmd_vel:=key_vel

expected:

- the **/scan_raw** topic should be available, **it is not there**
- topic **/head_front_camera/rgb/image_raw** is also missing

checked **so far**:

the **scan appears in the simulation window**, so it is active but **not producing any scan topic**

this is the laucher

```
import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import (DeclareLaunchArgument, GroupAction,
                            IncludeLaunchDescription, SetEnvironmentVariable)
from launch.conditions import IfCondition
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration, PythonExpression

def generate_launch_description():
    tiago_gazebo_dir = get_package_share_directory('tiago_gazebo')

    world = LaunchConfiguration('world', default='home')
    declare_world_cmd = DeclareLaunchArgument(
        'world',
        default_value='home',
        description='World name')

    tiago_sim_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(os.path.join(tiago_gazebo_dir, 'launch', 'tiago_gazebo.launch.py')),
        launch_arguments={
          'world_name': world
        }.items())

    declare_use_sim_time_cmd = DeclareLaunchArgument(
        'use_sim_time',
        default_value='True',
        description='Use simulation (Gazebo) clock if true')

    ld = LaunchDescription()

    ld.add_action(tiago_sim_cmd)

    ld.add_action(declare_use_sim_time_cmd)
    ld.add_action(declare_world_cmd)

    return ld

phil@ROS2:~$ find . -name tiago_gazebo.launch.py
./bookros2_ws/src/ThirdParty/tiago_simulation/tiago_gazebo/launch/tiago_gazebo.launch.py
```

then I **searched** for that **/scan_raw topic**

    phil@ROS2:~/bookros2_ws/src$ grep -r scan_raw . 
    ./ThirdParty/pmb2_robot/pmb2_description/urdf/base/base_sensors.urdf.xacro:      <xacro:hokuyo_urg_04lx_ug01_laser name="${name}_laser" parent="${name}" ros_topic="scan_raw" update_rate="10" min_angle="${-110 * deg_to_rad}" max_angle="${110 * deg_to_rad}" >
    ./ThirdParty/pmb2_robot/pmb2_description/urdf/base/base_sensors.urdf.xacro:      <xacro:sick_tim551_laser name="${name}_laser" parent="${name}" ros_topic="scan_raw" update_rate="10" min_angle="${-110 * deg_to_rad}" max_angle="${110 * deg_to_rad}" >
    ./ThirdParty/pmb2_robot/pmb2_description/urdf/base/base_sensors.urdf.xacro:      <xacro:sick_tim561_laser name="${name}_laser" parent="${name}" ros_topic="scan_raw" update_rate="10" min_angle="${-110 * deg_to_rad}" max_angle="${110 * deg_to_rad}" >
    ./ThirdParty/pmb2_robot/pmb2_description/urdf/base/base_sensors.urdf.xacro:      <xacro:sick_tim571_laser name="${name}_laser" parent="${name}" ros_topic="scan_raw" update_rate="10" min_angle="${-110 * deg_to_rad}" max_angle="${110 * deg_to_rad}" >
    ./ThirdParty/pmb2_robot/pmb2_description/CHANGELOG.rst:* scan_raw is the default laser topic
    ./ThirdParty/tiago_simulation/tiago_2dnav_gazebo/CHANGELOG.rst:  now uses laser filter to provide scan and scan_raw
    ./ThirdParty/tiago_simulation/tiago_2dnav_gazebo/CHANGELOG.rst:* now uses laser filter to provide scan and scan_raw
    ./ThirdParty/pmb2_navigation/pmb2_2dnav/CHANGELOG.rst:* use scan_raw for mapping
    ./ThirdParty/pmb2_navigation/pmb2_2dnav/params/pmb2_params.yaml:    scan_topic: /scan_raw
    ./ThirdParty/pmb2_navigation/pmb2_2dnav/params/pmb2_params.yaml:          topic: /scan_raw
    ./ThirdParty/pmb2_navigation/pmb2_2dnav/params/pmb2_params.yaml:          topic: /scan_raw
    ./ThirdParty/pmb2_navigation/pmb2_2dnav/params/pmb2_params.yaml:    scan_topic: /scan_raw
    ./ThirdParty/pmb2_navigation/pmb2_laser_sensors/launch/base_laser.launch:    <remap from="scan"          to="scan_raw"/>
    ./ThirdParty/pmb2_navigation/pmb2_laser_sensors/launch/hokuyo_laser.launch:    <remap from="scan" to="scan_raw"/>
    ./ThirdParty/pmb2_navigation/pmb2_laser_sensors/launch/sick-571_laser.launch:    <remap from="scan" to="scan_raw"/>
    ./ThirdParty/pmb2_navigation/pmb2_laser_sensors/launch/sick-551_laser.launch:    <remap from="scan" to="scan_raw"/>
    ./ThirdParty/pmb2_navigation/pmb2_laser_sensors/launch/sick-561_laser.launch:    <remap from="scan" to="scan_raw"/>
    ./ThirdParty/pal_gazebo_worlds/worlds/calibration_room.world:            <topicName>scan_raw</topicName>
    ./book_ros2/br2_fsm_bumpgo_cpp/launch/bump_and_go.launch.py:                        ('input_scan', '/scan_raw'),
    ./book_ros2/br2_bt_patrolling/launch/patrolling.launch.py:          ('input_scan', '/scan_raw'),
    ./book_ros2/br2_navigation/params/mapper_params_online_async.yaml:    scan_topic: /scan_raw
    ./book_ros2/br2_navigation/params/tiago_nav_params.yaml:    scan_topic: scan_raw
    ./book_ros2/br2_navigation/params/tiago_nav_params.yaml:          topic: /scan_raw
    ./book_ros2/br2_navigation/params/tiago_nav_params.yaml:          topic: /scan_raw
    ./book_ros2/br2_vff_avoidance/launch/avoidance_vff.launch.py:        ('input_scan', '/scan_raw'),
    ./book_ros2/br2_tf2_detector/launch/detector_basic.launch.py:                          ('input_scan', '/scan_raw')
    ./book_ros2/br2_tf2_detector/launch/detector_improved.launch.py:                          ('input_scan', '/scan_raw')
    ./book_ros2/br2_fsm_bumpgo_py/launch/bump_and_go.launch.py:                        ('input_scan', '/scan_raw'),
    ./book_ros2/code_annex.tex:                        ('input_scan', '/scan_raw'),
    ./book_ros2/code_annex.tex:                        ('input_scan', '/scan_raw'),
    ./book_ros2/code_annex.tex:                          ('input_scan', '/scan_raw')
    ./book_ros2/code_annex.tex:                          ('input_scan', '/scan_raw')
    ./book_ros2/code_annex.tex:        ('input_scan', '/scan_raw'),
    ./book_ros2/code_annex.tex:          ('input_scan', '/scan_raw'),
    ./install/br2_fsm_bumpgo_py/share/br2_fsm_bumpgo_py/launch/bump_and_go.launch.py:                        ('input_scan', '/scan_raw'),

I found this

**tiago_2dnav_gazebo/CHANGELOG.rst:  now uses laser filter to provide scan and scan_raw**

- **how do I use filters** ?? why is it not enabled by default ? :
- what could I check ?
- how to **activate a /scan_row topic** ?

thanks for your help

...and let me know if you need more information's

more details can be found on this post https://robotics.stackexchange.com/posts/104030/edit

saikishor, 2023-09-13:
Hello @phil123456,

Thank you for showing interest in our simulation. We will try to reproduce the issue at our end. We will update you on this soon.

Thank you,

Best Regards,
PAL Robotics Team.

Noel215, 2023-09-14:
Hello @phil123456,

We've been trying to reproduce the issue by using the same command as you specified:
```
#Terminal 1
ros2 launch br2_tiago sim.launch.py 
```

Noel215, 2023-09-14:
Hello @phil123456,

We've been trying to reproduce the issue by using the same commands as you specified:
```
# Terminal 1
ros2 launch br2_tiago sim.launch.py 

# Terminal 2
ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -r cmd_vel:=key_vel
```

However, we can see the topics that you say that are missing:
```
~/tmp_ws$ ros2 topic list  | grep raw
/head_front_camera/depth_registered/image_raw
/head_front_camera/rgb/image_raw
/scan_raw
```

> what could I check ?
> how to activate a /scan_row topic ?

Please, check that you are using the last version of the packages and try it in a clean workspace

> how do I use filters ?? why is it not enabled by default ? :

The laser filters are enabled for navigation, so if you want to use it you have to launch also navigation. You can use:
`ros2 launch br2_tiago sim.launch.py navigation:=true`

closed 2023-10-10 by saikishor