pal-robotics/tiago_simulation #27: Migrate to modern Gazebo

azeey, 2023-12-22:
As you may know, Gazebo-classic (aka Gazebo11, see [Terminology](https://gazebosim.org/about#:~:text=Terminology)) is no longer being actively developed  and will not be available on ROS Jazzy ([REP 2000](https://www.ros.org/reps/rep-2000.html#jazzy-jalisco-may-2024-may-2029)). I noticed that this project still uses Gazebo-classic. Are there any plans to migrate to modern Gazebo ([previously known as Ignition](https://community.gazebosim.org/t/a-new-era-for-gazebo/1356))?

These might be helpful resources if you do plan to migrate:

- Migration guide for SDF files: https://gazebosim.org/api/sim/8/migrationsdf.html 
- Migration guild for Gazebo plugins: https://gazebosim.org/api/sim/8/migrationplugins.html 
- Documentation for ros_gz: https://gazebosim.org/docs/harmonic/ros2_integration. 
- https://github.com/gazebosim/ros_gz_project_template and its [documentation](https://gazebosim.org/docs/harmonic/ros_gz_project_template_guide#guide-to-code-class-codespan-ros_gz_project_template-code-for-ros-2-and-gazebo-development).

jordan-palacios, 2024-02-06:
Hi @azeey.

Yes, we will be migrating the TIAGo simulation to supported versions of Gazebo in the future.

Thanks for your resouces.

EGAlberts, 2024-07-19:
Considering this is marked as closed, is the migration complete/available somewhere? @jordan-palacios ?

Rezenders, 2024-07-25:
I am also interested in this. Any updates?

saikishor, 2024-07-25:
Hello!

We do accept PRs. If any of you are interested and could help with the migration. We would be happy to review and get it merged!

Thank you!

Best Regards,
PAL Robotics Team

EGAlberts, 2024-07-25:
I think in that case this issue should be reopened right ;)

saikishor, 2024-07-25:
Hello @Rezenders and @EGAlberts ,

Thank you for taking time and working on this. To which of you both, should this issue be assigned?

Thank you, 

Best Regards,
PAL Robotics Team.

EGAlberts, 2024-07-31:
Hi @saikishor , I've begun the process with a series of forks including this main one: https://github.com/EGAlberts/tiago_simulation . I have gotten to the point of spawning tiago in gazebo fortress but without any plugins yet. Will maintain the progress in the forks and likely open issues in the specific repositories which might require changes.

dheerubhai-101, 2025-03-15:
hi, is there any update on this migration?

EGAlberts, 2025-03-19:
There was a complete migration made for Rolling And Jazzy (https://github.com/Tiago-Harmonic)