pal-robotics/tiago_simulation #28: Tiago Simulation Controllers not working in ROS 2 Humble and gazebo

nilutpolkashyap, 2024-01-13:
### Issue: Tiago Simulation ROS 2 humble branch does not move any of its joints. 

I was previously able to solve this issue here by just cloning the ros2_control humble branch and gazebo_ros2_control humble branch in my workspace and building them. Here is my previous issue #22 

But now, I am repeating the same steps but now the controllers are not working. I am not able to control the simulation using keyboard.

**System Specs**
1. Ubuntu 22.04
2. Gazebo Classic 11
3. ROS 2 Humble

**Steps I Followed**
1. Cloned [tiago_simulation](https://github.com/pal-robotics/tiago_simulation/tree/humble-devel) humble branch
2. git clone https://github.com/ros-controls/ros2_control.git -b humble 
3. git clone https://github.com/ros-controls/gazebo_ros2_control.git -b humble
4. git clone https://github.com/pal-robotics/pal_hey5.git -b humble-devel
5. git clone https://github.com/ros-perception/vision_msgs.git -b ros2
6. git clone https://github.com/pal-robotics/play_motion2.git -b humble-devel
7. git clone https://github.com/PickNikRobotics/launch_param_builder.git 
8. git clone https://github.com/pal-robotics/pal_urdf_utils.git -b humble-devel

I am trying to launch Tiago robot simulation in Gazebo Classic using 
```ros2 launch tiago_gazebo tiago_gazebo.launch.py arm:=no-arm```

Terminal Output (with Error)
```
[robot_state_publisher-12] [INFO] [1705147635.308914566] [robot_state_publisher]: got segment caster_front_left_2_link
[robot_state_publisher-12] [INFO] [1705147635.308925948] [robot_state_publisher]: got segment caster_front_right_1_link
[robot_state_publisher-12] [INFO] [1705147635.308941278] [robot_state_publisher]: got segment caster_front_right_2_link
[robot_state_publisher-12] [INFO] [1705147635.308953569] [robot_state_publisher]: got segment head_1_link
[robot_state_publisher-12] [INFO] [1705147635.308966719] [robot_state_publisher]: got segment head_2_link
[robot_state_publisher-12] [INFO] [1705147635.308978541] [robot_state_publisher]: got segment head_front_camera_depth_frame
[robot_state_publisher-12] [INFO] [1705147635.308990281] [robot_state_publisher]: got segment head_front_camera_depth_optical_frame
[robot_state_publisher-12] [INFO] [1705147635.309004789] [robot_state_publisher]: got segment head_front_camera_link
[robot_state_publisher-12] [INFO] [1705147635.309016572] [robot_state_publisher]: got segment head_front_camera_optical_frame
[robot_state_publisher-12] [INFO] [1705147635.309028213] [robot_state_publisher]: got segment head_front_camera_orbbec_aux_joint_frame
[robot_state_publisher-12] [INFO] [1705147635.309039775] [robot_state_publisher]: got segment head_front_camera_rgb_frame
[robot_state_publisher-12] [INFO] [1705147635.309051434] [robot_state_publisher]: got segment head_front_camera_rgb_optical_frame
[robot_state_publisher-12] [INFO] [1705147635.309062939] [robot_state_publisher]: got segment rgbd_laser_link
[robot_state_publisher-12] [INFO] [1705147635.309074355] [robot_state_publisher]: got segment suspension_left_link
[robot_state_publisher-12] [INFO] [1705147635.309085796] [robot_state_publisher]: got segment suspension_right_link
[robot_state_publisher-12] [INFO] [1705147635.309097577] [robot_state_publisher]: got segment torso_fixed_column_link
[robot_state_publisher-12] [INFO] [1705147635.309108986] [robot_state_publisher]: got segment torso_fixed_link
[robot_state_publisher-12] [INFO] [1705147635.309120497] [robot_state_publisher]: got segment torso_lift_link
[robot_state_publisher-12] [INFO] [1705147635.309131912] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-12] [INFO] [1705147635.309143308] [robot_state_publisher]: got segment wheel_right_link
[spawn_entity.py-3] [INFO] [1705147636.134563543] [spawn_entity]: Spawn Entity started
[spawn_entity.py-3] [INFO] [1705147636.135440982] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-3] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-3]   warnings.warn(
[spawn_entity.py-3] [INFO] [1705147636.440675382] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-3] [INFO] [1705147636.638303328] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-3] [INFO] [1705147636.639161423] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-3] [INFO] [1705147637.647572551] [spawn_entity]: Calling service /spawn_entity
[spawner-5] [INFO] [1705147637.837050264] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1705147638.075292623] [spawner_head_controller]: Waiting for '/controller_manager' node to exist
[gzserver-1] [INFO] [1705147638.322908527] [base_imu]: <initial_orientation_as_reference> is unset, using default value of false to comply with REP 145 (world as orientation reference)
[spawner-6] [INFO] [1705147638.621901077] [spawner_torso_controller]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1705147638.689680166] [spawner_mobile_base_controller]: Waiting for '/controller_manager' node to exist
[spawn_entity.py-3] [INFO] [1705147638.860202921] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [tiago]
[INFO] [spawn_entity.py-3]: process has finished cleanly [pid 19506]
[gzserver-1] [INFO] [1705147639.711306066] [head_front_camera_frame_controller]: Publishing camera info to [/head_front_camera/rgb/camera_info]
[gzserver-1] [INFO] [1705147639.713488610] [head_front_camera_frame_controller]: Publishing depth camera info to [/head_front_camera/depth_registered/camera_info]
[gzserver-1] [INFO] [1705147639.715055198] [head_front_camera_frame_controller]: Publishing pointcloud to [/head_front_camera/depth_registered/points]
[spawner-5] [INFO] [1705147639.859204517] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[gzserver-1] [INFO] [1705147639.867468382] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1705147639.895916386] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gzserver-1] [INFO] [1705147639.896004483] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1705147639.899481841] [gazebo_ros2_control]: Loading parameter files /home/ubuntu/bookros2_ws/install/tiago_controller_configuration/share/tiago_controller_configuration/config/gazebo_controller_manager_cfg.yaml
[gzserver-1] [INFO] [1705147639.926343223] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[... 96 lines cut ...]
[tuck_arm.py-13] [ERROR] [1705147668.869709813] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-8] [ERROR] [1705147672.873548943] [play_motion2]: Service /controller_manager/list_controllers not available.
[play_motion2_node-8] [ERROR] [1705147672.873652845] [play_motion2]: There are no active JointTrajectory controllers available
[tuck_arm.py-13] [ERROR] [1705147672.875283442] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-8] [ERROR] [1705147676.878690298] [play_motion2]: Service /controller_manager/list_controllers not available.
[play_motion2_node-8] [ERROR] [1705147676.878863132] [play_motion2]: There are no active JointTrajectory controllers available
[tuck_arm.py-13] [ERROR] [1705147676.885502266] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-8] [ERROR] [1705147680.893137080] [play_motion2]: Service /controller_manager/list_controllers not available.
[play_motion2_node-8] [ERROR] [1705147680.893235422] [play_motion2]: There are no active JointTrajectory controllers available
[tuck_arm.py-13] [ERROR] [1705147680.894898665] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-8] [ERROR] [1705147684.903039343] [play_motion2]: Service /controller_manager/list_controllers not available.
[play_motion2_node-8] [ERROR] [1705147684.903175367] [play_motion2]: There are no active JointTrajectory controllers available
[tuck_arm.py-13] [ERROR] [1705147684.904305059] [arm_tucker]: play_motion2 is not ready
```

[image]

[image]

Even if I try to move the robot, I get nothing, no movement
```ros2 topic pub /mobile_base_controller/cmd_vel_unstamped geometry_msgs/msg/Twist '{linear: {x: 1}, angular: {z: 0}}' -r10```

[image]

Please help me out. Thank you

nilutpolkashyap, 2024-01-16:
Hi @Noel215 , any help would be appreciated

Noel215, 2024-01-16:
Hi @nilutpolkashyap,

It stopped working after they merged https://github.com/ros-controls/ros2_control/pull/1272 4 days ago.
I just opened an issue in the `ros2_control` repository.

For now you can checkout your local repository to the latest tag to make it work
```
git checkout 2.36.1
```

I hope that resolves your problem.

nilutpolkashyap, 2024-01-17:
Hi @Noel215 
I checkout out version 2.36.1 but I am again getting the following error. 
[image]
[image]

I also tried out different versions of ```ros2_control``` and ```gazebo_ros2_control``` but I still get the same error

saikishor, 2024-01-24:
hello!

It will be solved with the upcoming sync: https://discourse.ros.org/t/preparing-for-humble-sync-and-patch-release-2024-01-26/35765

Thank you for your patience

Best Regards,
PAL Robotics Team

gstavrinos, 2024-02-07:
Just as a heads-up, I am experiencing the same issue with all the latest releases (checkout from source ros2_controllers, gazebo_ros2_control, ros2_control along with all the latest humble releases of your packages).

Any insight would be highly appreciated!

saikishor, 2024-02-07:
Hello @gstavrinos !

This has been recently reviewed and fixed. It should work. If it isn't, please send us more information to reproduce it. Try with the binaries installed, instead of compiling from source. We hope it helps.

Best regards,
PAL Robotics Team

gstavrinos, 2024-02-07:
Hey @saikishor,

Thank you for your quick reply. I tried both with the binaries and from source with the exact same results. I can provide you with whatever information you may ask, but I basically face the exact same error specified in this issue (missing controller manager).

saikishor, 2024-02-07:
Thanks for the prompt reply.  Can you please share the versions of the above packages and also hardware_interface and controller_mansger?

Thank you,

Best regards,
PAL Robotics Team

gstavrinos, 2024-02-07:
Here you are:

* `gazebo_ros2_control`: `0.4.6`
* `hardware_interface`: `3.28.0`
* `ros2_controllers`: `2.32.0`
* `ros2_control`: `2.38.0`
---
* `tiago_moveit_config`: `3.0.9`
* `tiago_navigation`: `4.0.10`
* `tiago_gazebo`: `4.0.9`
* `tiago_robot`: `4.2.2`
---
* `play_motion2`: `0.0.15`
* `launch_pal`: `0.0.18`
---
* `pal_gazebo_worlds`: `4.0.3`
* `pal_urdf_utils`: `2.0.1`
---

I understand that some of there are most probably irrelevant, but I guess "excessive" information would not hurt!

Noel215, 2024-02-12:
Hi @gstavrinos,

The latest released versions are working, you don't need to have a workspace.
Just open a new environment (e.g. docker) and run:
`sudo apt update`
`sudo apt install ros-humble-tiago-simulation`

In case you want to have it in your workspace here you have the latest release for each repository on `humble`:

---
- `ros2_control`: `2.37.0`
- `ros2_controllers`: `2.32.0`
- `gazebo_ros2_control`: `0.4.6`
---
- `tiago_moveit_config`: `3.0.7`
- `tiago_nav`: `4.0.9`
- `tiago_robot`: `4.1.2`
- `tiago_simulation`: `4.0.8`
---
- `play_motion2`: `0.0.13`
- `launch_pal`: `0.0.16`
---
- `pal_gazebo_worlds`: `4.0.3`
- `pal_urdf_utils`: `2.0.1`
---

Let us know, please.

nilutpolkashyap, 2024-02-12:
Hi @Noel215 , 
Your previous solutions helped me to solve the robot's control problem with teleop. I was able to drive the robot. 

But today I checked the robot's head joints and also tried to use the moveit demos as shown [here](https://github.com/pal-robotics/tiago_simulation?tab=readme-ov-file#simulation--moveit-2) , none of the joints worked. I am not able to control any of the joints. When I try to execute the trajectory, the Moveit example shows me a "Failed" message in Rviz2.

Any help would be appreciated

Noel215, 2024-02-13:
Hi @nilutpolkashyap,

I've just tested it in a fresh `humble` docker with the debian installation and everything is working for me:
```
sudo apt install ros-humble-tiago-simulation
```
I've tried with some random valid trajectories.

Please, try it in the same way and if it's still not working give us more details in order to reproduce the issue.

gstavrinos, 2024-02-20:
I am getting back to this, just to confirm that, indeed, a completely fresh installation of the binary packages works without problems. Thank you!

saikishor, 2024-02-20:
Thank you for confirming it. I'll go ahead and close the issue

closed 2024-02-20 by saikishor