pal-robotics/tiago_simulation #29: Demos not working

adamkrawczyk, 2024-02-03:
Hi, 
I have tried to launch the simulation but I have found the following issues: 

1. `ros2 launch tiago_2dnav tiago_nav_bringup.launch.py is_public_sim:=true` :
```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): launch configuration 'world_name' does not exist
```
2. `ros2 launch tiago_gazebo tiago_gazebo.launch.py navigation:=true is_public_sim:=true [arm:=no-arm]`
```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'pal_nav2_bringup' not found
```

I can't find the repo for the `pal_nav2_bringup` seems that it's private or not published yet. Can't find this file in .repos (btw strange that this file is not added to the top level of the repo - could be issues with versions) . 

Could you suggest some ways to resolve that? 

Best

Noel215, 2024-02-13:
Hi @adamkrawczyk, 

We've done some fixes, could you try again?

If you want to be sure that everything works, you can use the debian installation:
```
sudo apt install ros-humble-tiago-simulation
```
Keep in mind that if you're using the latest version of the repositories, there might be ongoing changes, and some functionalities may not work as expected.

Noel215, 2024-02-13:
The changes have been made on `tiago_navigation` and `pmb2_navigation` repositories, not on this one. Please, check them out.

I can reproduce the laser filter issue, we'll take a look on that.

kavikode, 2024-02-24:
I received the following error message: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

Can you please help to fix this issue?

Noel215, 2024-03-01:
Hi @kavikode, @adamkrawczyk,

I've just released the latest version of some packages, in order to update them you can use `vcs pull`.

Note that MoveIt 2 is launched now by default, so in order to launch only navigation you will need to run:

```
ros2 launch tiago_gazebo tiago_gazebo.launch.py moveit:=false navigation:=true is_public_sim:=true world_name:=pal_office
```

Don't worry if you have some error logs of `play_motion2`. It needs MoveIt 2, so it's normal.

Everything should be working with the latest version, so I'm closing this issue. 
Don't hesitate to comment again if you're still experiencing the same issues, or feel free to open a new issue if it's unrelated.

david-wb, 2024-05-27:
@Noel215 This is still an issue for me after doing a `vcs pull`. The error is `package 'pal_nav2_bringup' not found`:

```
ros2 launch tiago_gazebo tiago_gazebo.launch.py navigation:=True is_public_sim:=True arm:=no-arm
[INFO] [launch]: All log files can be found below /home/chroma/.ros/log/2024-05-27-13-53-00-342308-chroma-HP-ProBook-650-G2-163103
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'pal_nav2_bringup' not found, searching: ['/home/chroma/tiago_public_ws/install/urdf_test', '/home/chroma/tiago_public_ws/install/tiago_simulation', '/home/chroma/tiago_public_ws/install/tiago_robot', '/home/chroma/tiago_public_ws/install/tiago_navigation', '/home/chroma/tiago_public_ws/install/tiago_gazebo', '/home/chroma/tiago_public_ws/install/tiago_moveit_config', '/home/chroma/tiago_public_ws/install/tiago_laser_sensors', '/home/chroma/tiago_public_ws/install/tiago_bringup', '/home/chroma/tiago_public_ws/install/tiago_description', '/home/chroma/tiago_public_ws/install/tiago_controller_configuration', '/home/chroma/tiago_public_ws/install/tiago_2dnav', '/home/chroma/tiago_public_ws/install/pmb2_robot', '/home/chroma/tiago_public_ws/install/pmb2_navigation', '/home/chroma/tiago_public_ws/install/pmb2_2dnav', '/home/chroma/tiago_public_ws/install/pmb2_laser_sensors', '/home/chroma/tiago_public_ws/install/pmb2_bringup', '/home/chroma/tiago_public_ws/install/pmb2_description', '/home/chroma/tiago_public_ws/install/pmb2_controller_configuration', '/home/chroma/tiago_public_ws/install/play_motion2', '/home/chroma/tiago_public_ws/install/play_motion2_msgs', '/home/chroma/tiago_public_ws/install/pal_hey5', '/home/chroma/tiago_public_ws/install/pal_hey5_description', '/home/chroma/tiago_public_ws/install/pal_gripper_simulation', '/home/chroma/tiago_public_ws/install/pal_gripper', '/home/chroma/tiago_public_ws/install/pal_gripper_description', '/home/chroma/tiago_public_ws/install/pal_urdf_utils', '/home/chroma/tiago_public_ws/install/pal_robotiq_gripper', '/home/chroma/tiago_public_ws/install/pal_robotiq_description', '/home/chroma/tiago_public_ws/install/pal_robotiq_controller_configuration', '/home/chroma/tiago_public_ws/install/pal_navigation_cfg', '/home/chroma/tiago_public_ws/install/pal_navigation_cfg_bringup', '/home/chroma/tiago_public_ws/install/pal_navigation_cfg_params', '/home/chroma/tiago_public_ws/install/pal_hey5_controller_configuration', '/home/chroma/tiago_public_ws/install/pal_gripper_controller_configuration', '/home/chroma/tiago_public_ws/install/pal_gazebo_worlds', '/home/chroma/tiago_public_ws/install/launch_pal', '/opt/ros/humble']"
```

Noel215, 2024-05-29:
Hi @david-wb,

We are looking into it, sorry for the inconvinience.

Noel215, 2024-05-29:
It should be fixed now, you can update the packages by running `vcs pull`.

closed 2024-05-29 by Noel215