pal-robotics/tiago_simulation #34: tiago_sim_nav_bringup.launch.py not found

david-wb, 2024-05-27:
When trying to launch the nav2 example, I get the following error.
```
$ ros2 launch tiago_gazebo tiago_gazebo.launch.py navigation:=True is_public_sim:=True arm:=no-arm world_name:=small_office
[INFO] [launch]: All log files can be found below /home/chroma/.ros/log/2024-05-27-14-27-15-117875-chroma-HP-ProBook-650-G2-5881
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): [Errno 2] No such file or directory: '/home/chroma/tiago_public_ws/install/tiago_2dnav/share/tiago_2dnav/launch/tiago_sim_nav_bringup.launch.py'
```
Where is the missing launch file `tiago_sim_nav_bringup.launch.py`? Thanks for your assistance.

Noel215, 2024-05-29:
Hi @david-wb,

That launcher was removed some time ago. We've recently restructured the launch files, we have missed that change.
Thanks for reporting it, I've left you a comment in the PR.

kavikode, 2024-06-02:
How do you rebase in the local repository these changes because VCS import does not use git? Please show the commands to update the local repository from https://github.com/david-wb/tiago_simulation.git tiago_simulation_custom.

I am only aware of using git clone --branch fix/tiago_nav_bringup https://github.com/david-wb/tiago_simulation.git tiago_simulation_custom

Noel215, 2024-06-03:
@kavikode it's merged, so you can use `vcs pull` to get the latest changes.

FYI you could have used `git remote add` to add a new remote repository to your local one. Then you can pull the changes.
See the [git documentation](https://www.git-scm.com/docs/git#_git_commands) for more information

closed 2024-06-03 by Noel215

kavikode, 2024-06-04:
thank you so much!