pal-robotics/tiago_simulation #36: Compressed Image Topic

kavikode, 2024-06-05:
I understand that this topic is available for the camera from /head_front_camera/rgb/image_raw
is there a compressed image topic?
Thank you

saikishor, 2024-06-05:
Hello!

Can you please give some more context on this?

Best Regards,
Sai Kishor Kothakota

kavikode, 2024-06-05:
sudo apt install ros-humble-image-transport-plugins ros-humble-theora-image-transport ros-humble-compressed-depth-image-transport ros-humble-compressed-image-transport

seem to solve the issue

After a showing the ros2 topic list, I get the following
---
/head_front_camera/depth/image_raw/compressed
/head_front_camera/depth/image_raw/compressedDepth
/head_front_camera/depth/image_raw/theora
/head_front_camera/depth_registered/camera_info
/head_front_camera/depth_registered/image_raw
/head_front_camera/depth_registered/points
/head_front_camera/image_raw/compressed
/head_front_camera/image_raw/compressedDepth
/head_front_camera/image_raw/theora
/head_front_camera/rgb/camera_info
/head_front_camera/rgb/image_raw
---

Thank you. Please include in the readme file.