pal-robotics/tiago_simulation #37: Map frame not published for Nav2

kbarwich, 2024-06-07:
Hi, Im having a bit of an issue getting Nav2 to work with the simulation
When following the README, trying to start the Simulation followed by Navigation2, no map shows up in RVIZ. While there is a /map topic published, the TF2 Frame does not seem to exist. 
Using ROS2 Humble, noticed no errors during the build process.

sudo-michael, 2024-07-24:
It appears that the `/odom` frame isn't being published. I went through the solutions [here](https://github.com/pal-robotics/tiago_simulation/issues/23) but I'm not sure why it's not working again.

dseredyn, 2024-07-27:
I've found how to fix it.
You need to change the file:
pmb2_robot/pmb2_controller_configuration/config/mobile_base_controller.yaml
change:
enable_odom_tf: false
into:
enable_odom_tf: true