pal-robotics/tiago_simulation #40: Converting xacro to urdf generate empty files

kavikode, 2024-08-12:
In ROS2 Humble, I execute the command:
find . -type f -name "*.urdf.xacro" -exec sh -c 'xacro "$0" > "$(dirname "$0")/$(basename "$0" .urdf.xacro).urdf"' {} \;

It only generated empty files. How can I correct this?

saikishor, 2024-08-13:
Hello @kavikode!

The URDF itself needs some parameters to generate the final consolidated xacro. Please take a look at the args and see how it is being generated.

Thank you!

Best Regards,
PAL Robotics Team

kavikode, 2024-08-14:
thank you for your message. I'm trying to do the opposite. I want to convert the xacro files into urdf files. 

I performed the following:  xacro tiago.urdf.xacro > tiago.urdf inside the folder
tiago_description/robots.

Then inside the folder tiago_description/urdf/arm,
I performed the following: 
xacro arm.urdf.xacro > arm.urdf
 
and it's empty.

The same results for all of the other joints of the robot. Please help.

kavikode, 2024-08-22:
Any other suggestions, please?

saikishor, 2024-08-22:
Hello @kavikode!

In order to get the final URDF, you will need to pass all the defined arguments. For instance, the following list of the arguments : https://github.com/pal-robotics/tiago_robot/blob/humble-devel/tiago_description/launch/robot_state_publisher.launch.py#L92-L114. If you need to do it per xacro of folder, it is might not be possible because they are xacro macros, so you won't be able to create directly, unless you write a separate .urdf.xacro that uses that macro

Thank you!

saikishor, 2024-09-02:
Hello @kavikode!

We hope the above solution fixed your problem. Closing the issue for now!

Thank you!

Best Regards,
PAL Robotics Team

closed 2024-09-02 by saikishor