pal-robotics/tiago_simulation #48: Kinetic Branch also compatible with ros noetic?

NamTruongTran, 2025-01-28:
Hey,

does the Kinetic Branch also work for ros noetic ?

Thang you !

Best regards,
Truong

saikishor, 2025-01-28:
Hello @NamTruongTran!

Yes, it should work with `noetic` as well!

NamTruongTran, 2025-01-28:
Thank you so much for you answer @saikishor.
I really appreciate your time :)

I have another question:
Does this repo also work for turtlebot3?:
https://github.com/pal-robotics/pal_person_detector_opencv

Best regards,
Truong

NamTruongTran, 2025-01-30:
Hey @saikishor ,

do you know how i can include my custom maps ?
I don't understand the folder structure with tiago_maps/config and tiago_mags/configuration. 
Also what does the transformation.xml file or the mmap.yaml do and how can i create it for my custom maps?

https://github.com/pal-robotics/tiago_navigation/tree/kinetic-devel/tiago_maps

Thank you :)

saikishor, 2025-01-30:
> I don't understand the folder structure with tiago_maps/config and tiago_mags/configuration.
> Also what does the transformation.xml file or the mmap.yaml do and how can i create it for my custom maps?

Hello @NamTruongTran!

The transformation.xml or mmap.yaml is created when you finish the mapping of the environment. One defines the center of the map or the starting point of the robot when creating the map. mmap.yaml is for some internal software that can make the robot navigate to the point of interest.

Thank you!

closed 2025-01-30 by saikishor