pal-robotics/tiago_simulation #49: Simulation problems after package binary installation

SeungRyeol, 2025-02-23:
Hello. After this deployment, after running the simulation with the command below, an error like the picture below occurs.

```bash
sudo apt install ros-humble-tiago-gazebo
ros2 launch tiago_gazebo tiago_gazebo.launch.py is_public_sim:=True
```

[image]

There was no such problem a few weeks ago, but there seems to be a problem with this distribution.

enrico391, 2025-03-17:
Make sure to have topic robot_description publishes over ros2 , try to use ros2 topic list and see if robot_description topic exists

rajRishi22, 2025-03-30:
Hi , I am also facing the same issue , I ran the command ros2 topic list and robot_description topic exists but still i am getting the error .

saikishor, 2025-03-30:
This is the issue from the MoveIt side. You have to use the play_motion2 locally. If not, you should wait for next Humble sync

rajRishi22, 2025-03-30:
Thanks for immediate response , i will use use play_motion2 locally then .

Noel215, 2025-05-13:
This was fixed on https://github.com/pal-robotics/play_motion2/commit/db4ad204cd3bacbac38bb76b14e86d046ae2a49c. It should work with the latest version of `humble`

closed 2025-05-13 by Noel215